shiyu1314/openwrt-onecloud #18: 能直接把固件扩容

dj56959566, 2024-10-31:
本身one cloud 1+8G 装完系统应当还有空间。
能把软件包大小调到2-3G  docker 用剩余空间呢。这样方便我们这些小白折腾。
试了好多方法都没有扩容成功。一扩就系统坏了。删除错了。谢谢作者！

dj56959566, 2024-10-31:
[image]
[image]
剩余7.2G 我分了二个区一个贸给了docker一个留给系统软件包。但系统软件包挂不上。是不是要删除一个分区才能持上去呀。

dj56959566, 2024-10-31:
[image]
[image]

shiyu1314, 2024-10-31:
使用resize2fs扩容确实有几率丢失数据，只要扩容后重启两次内核日志搜索ext4关键有6个信息就正常。没有就重刷，一般是线刷停了或继续导致

dj56959566, 2024-10-31:
能不有能直接做一个扩容的固件。软件包2-3G docker用剩余的呢。

shiyu1314, 2024-10-31:
刷好执行sh 1.sh

dj56959566, 2024-10-31:
不懂下次直接做分别做一个。我是人小白。哈哈。这情况弄了三天了。见笑了现在还没有弄好。现在docker又不行了。要得重装。

dj56959566, 2024-10-31:
自已会断点

adjlevis, 2024-11-01:
老大我今天换docker源又整一天，没换成功系统刷了n次……

closed 2024-12-16 by shiyu1314